zapmaker/GrblHoming #100: Not installing in ubuntu 14.04

arthurwolf, 2015-03-08:
Hey.

I ran : sudo apt-get install "^libxcb.*" libx11-xcb-dev libglu1-mesa-dev libxrender-dev 

Then downloaded and ran http://zapmaker.grblcontroller.s3-website-us-west-2.amazonaws.com/GrblControllerLinuxInstallerx64-3.6.1

And it gives me this error : 

# arthur at aquarelle in ~ [12:22:35]
$ sudo ./GrblControllerLinuxInstallerx64-3.6.1
./GrblControllerLinuxInstallerx64-3.6.1: error while loading shared libraries: libxcb-sync.so.0: cannot open shared object file: No such file or directory

I tried installing the missing library but it seems to be there already : 

# arthur at aquarelle in ~ [12:22:37]
$ bash                                        
arthur@aquarelle:~$ sudo apt-get install libxcb-sync*
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libxcb-sync0-dev' for regex 'libxcb-sync*'
Note, selecting 'libxcb-sync1-dbg' for regex 'libxcb-sync*'
Note, selecting 'libxcb-sync1' for regex 'libxcb-sync*'
Note, selecting 'libxcb-sync-dev' for regex 'libxcb-sync*'
Note, selecting 'libxcb-sync-dev' instead of 'libxcb-sync0-dev'
libxcb-sync-dev is already the newest version.
libxcb-sync1 is already the newest version.
libxcb-sync1-dbg is already the newest version.
The following packages were automatically installed and are no longer required:
  linux-headers-3.13.0-32 linux-headers-3.13.0-32-generic
  linux-headers-3.13.0-39 linux-headers-3.13.0-39-generic
  linux-image-3.13.0-32-generic linux-image-3.13.0-39-generic
  linux-image-extra-3.13.0-32-generic linux-image-extra-3.13.0-39-generic
  linux-image-generic
Use 'apt-get autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 141 not upgraded.
arthur@aquarelle:~$ 

I want to try this software to see if it works with the Smoothie firmware ( and if needed get it to work ).

Thanks for your help.

Cheers.

ChmlGr, 2015-09-25:
Same problem here

jaiagreen, 2015-11-15:
I'm also having this problem with Ubuntu 14.04.

smoksmk, 2016-07-23:
I install 
https://packages.debian.org/hu/wheezy/amd64/libxcb-sync0/download
and 
http://packages.ubuntu.com/ru/precise/amd64/libicu48/download
then worked

marcopoles, 2017-01-03:
Hi, after installing previous libs now I have following error:

Fatal: QFontDatabase: Cannot find font directory /usr/local/Qt-5.3.1/lib/fonts - is Qt installed correctly?
(fontdatabase/basic/qbasicfontdatabase.cpp:99)

I'm using Puppy Linux Tahrpup 6.05 (based on debian/ubuntu packages) and I installed Qt 5.5 with package tool. I see that /usr/bin/qtcreator is present but I don't able to find font directory. 
I tried also with Lubuntu, but I had same error 

Did you had a similar problem?

Thanks,
Marco

treki1, 2017-02-04:
~/downloads $ sudo ./GrblControllerLinuxInstallerx64-3.6.1
sudo: unable to execute ./GrblControllerLinuxInstallerx64-3.6.1: Permission denied
The same problem as root.
My System: Linux Mint 18.1 mate amd64
Any advice?

BenBlumer, 2017-11-12:
@smoksmk  packages solved the problem. Though, his link for libicu48 is dead. So I use this [one instead](https://packages.debian.org/wheezy/amd64/libicu48/download).