zapmaker/GrblHoming #102: Choose File doesn't work

lawmailroomcom, 2015-05-20:
When I select "Choose File", the cursor turns into a double headed arrow.  I expected a file browser window.  Is this what is supposed to happen?

jshaw, 2015-05-27:
I am using Grbl Controller 3.5 on OS X and I am running into the same issue. When I select to choose a file I get a double arrow as my cursor but no file dialogue. I have found that if I type in a correct path I still can get access to the file for my machine but this still looks like a bug in the functionality.

It also seems to be a related issue but when I add the path to my .nc file i do not see the GCode visualization in the visualizer tab on the right... I can cut with the path I put in the send file input box I just cant seem to see it before I start the cutting.

I am on OS X Yosemite 10.10.3.

guerrilladrone, 2015-08-18:
were you guys able to find a solution to this issue? I updated my macbook and after that my grbl does the same as you guy's.... any help?

Thanks!

Uiffie, 2015-08-26:
same problem here!
I updated to osx 10.10.5. I can't open files and enjoy a double headed arrow.
anybody? Thank you!

thanhmeye, 2015-09-01:
I'm running grbl 3.5 on Yosemite to control my Shapeoko 2. Grbl allows me to connect to the machine, jog it in xyz directions and set Zero position. However, when I clicked on Choose file, nothing comes up that would allow browsing .nc file. I used to run an old version of Grbl on OSX Lion about 9 months back, and it worked fine. That was before upgrading to Yosemite.

Until this issue is fixed, I will downgrade to an older grbl version, hopefully that would work.

pspavlov, 2015-09-19:
Same here . Just updated  OS X to Yosemite 10.10.5  and now chose file button does not work. The cursor changes to a "resize" arrow but no UI appears to select a file.