zapmaker/GrblHoming #108: download link is not working

ayfati01, 2015-11-05:
guys ive tried to download grbl controller but whenever i click on the given link above, a 404 page pops up all the time. pleeeeeeeeease help me.

kurumaz, 2015-11-06:
same problem over here!

abdoale, 2015-11-11:
guys ive tried to download grbl controller but whenever i click on the given link above, a 404 page pops up all the time. pleeeeeeeeease help me.

Cvcolod, 2015-11-12:
Same here, Does someone fix it ?

LETARTARE, 2015-11-12:
I have no problem with the download button 'DownLoad Zip ' (Vista).

internetcobus, 2015-11-13:
Same here, cannot download. I get a 403 Forbidden page.

tobycatlin, 2015-11-20:
I have attached a zip file containing the installer. I had to rename the zip to text but renaming back again should allow you to unpack it.

[GrblController361Setup.txt](https://github.com/zapmaker/GrblHoming/files/40240/GrblController361Setup.txt)

ayfati01, 2015-11-21:
i tried to make it a zip file but the computer said that there is a problem
with the file.

internetcobus, 2015-11-23:
Thanks, it worked for me. Just download and renamed to "GrblController361Setup.zip".

kurumaz, 2015-11-23:
It worked for me too. Tks!

Reissuli, 2015-12-09:
Could I get the download link for mac installer please?

devolf92, 2016-02-09:
still got 403 error, from ukrainian and american ip.
thanks for txt!

LittleMiller4, 2016-12-06:
Google Chrome detected the text download is a virus. Thanks, or no thanks?

LittleMiller4, 2016-12-06:
Here is the download link for the grbl installer: https://github.com/zapmaker/GrblHoming/releases