zapmaker/GrblHoming #118: Error while loading shared libraries: libxcb-sync.so.0

space55, 2017-03-03:
Hello

I'm having an issue on Debian 8 x64 with GrblController. I try to run the installer, however, it gives me the following error:
`./GrblControllerLinuxInstallerx64-3.6.1: error while loading shared libraries: libxcb-sync.so.0: cannot open shared object file: No such file or directory`

I've tried the fixes listed in #100 and they don't appear to be working. Am I doing something wrong, or is there some sort of library issue?

Thanks,
Space

IgniparousTempest, 2017-09-24:
Do you have a solution to this?

icastillejogomez, 2018-03-15:
I have the same error in Ubuntu 16.04.4 LTS  :(

`./GrblControllerLinuxInstallerx64-3.6.1: error while loading shared libraries: libxcb-sync.so.0: cannot open shared object file: No such file or directory`

HenrikJo, 2018-08-16:
Same here, any luck?

space55, 2018-09-04:
@IgniparousTempest @icastillejogomez @HenrikJo The only solution I've found is to run a 32-bit Debian VM. Maybe a Docker container would work, but I haven't tried it.

icastillejogomez, 2018-09-05:
@space55 yes, maybe running on docker container works. Thanks for reply

aurelio0rs1araujo, 2019-08-12:
excute this command: 
cd /usr/lib/x86_64-linux-gnu
ln -s libxcb-sync.so.1.0.0 libxcb-sync.so.0